Edzelf/ESP32-Radio #502: Webserver stops responding

arnolde, 2022-03-23:
Thank you for this fantastic project! I had a big problem when my Dad became senile and wasnt able to operate his TV or radio anymore. He lives in an area with bad radio reception and no satellite receiver was simple enough for him to use, he started going crazy if he couldnt listen to classical music all the time. So I built him a minimal radio with no controls or display, just an ESP32 and VS1053 module, power in, headphones out, one single station permanently configured, viola. It's been a great help for the past 2 weeks, thank you so much!

Now my problem: It seems that after a few days (?) of operation, the ESP32 webserver stops working. The radio is still playing, the internet router still sees the ESP connected and producing traffic, but accessing the config webpage times out. After a powercycle it works again. I'm sorry I can't debug it any more than that, since it's in a location remote to me. It's not a serious problem, but I thought I'd mention it, maybe someone else has also observed it?

Edzelf, 2022-03-23:
Maybe switching to version 2 helps.

xP373Rx, 2022-04-22:
There is something weird gouing on with the esp32 connect  function  i replaced  
if ( mp3client.connect ( hostwoext.c_str(), port ) )

with
if ( mp3client.connect ( hostwoext.c_str(), port,5000 ) )    //In function connecttohost()

That one has a timeout since the  one without timeout parameter is blocking my http pages sometimes ..
Now there is no blocking on dead urls.