Edzelf/ESP32-Radio #505: SD card is read, file count correct but no way to play them

Fabri54, 2022-04-23:
After a few days scratching my head I had t ask for help. I've successfully assembled the Internet radio, with a red TFT display (+SD card).
Compilation is fine as well as download to the ESP32 and I can enjoy the music of internet radios. However what doesn't work is the SD card. The card is seen at start-up, the number of track is correct but there is no way to have them played from web interface.
 If I go to MP3 Player on the web page the re are no tracks to play. When I chose RANDOM the system crashes and reboots

Any help would be greatly appreciated. Thank you!

Here is the initial diagnostic at boot.

20:56:19.420 -> mode:DIO, clock div:1
20:56:19.457 -> load:0x3fff0018,len:4
20:56:19.457 -> load:0x3fff001c,len:1216
20:56:19.457 -> ho 0 tail 12 room 4
20:56:19.457 -> load:0x40078000,len:10944
20:56:19.457 -> load:0x40080400,len:6388
20:56:19.457 -> entry 0x400806b4
20:56:19.753 -> 
20:56:19.753 -> D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Mon, 28 Jun 2021 12:40:00 GMT.  Free memory 301700
20:56:19.787 -> D: Display type is BLUETFT
20:56:19.787 -> D: Partition nvs found, 20480 bytes
20:56:19.787 -> D: Read 43 keys from NVS
20:56:19.787 -> D: pin_ir set to 35
20:56:19.787 -> D: pin_enc_clk set to 25
20:56:19.787 -> D: pin_enc_dt set to 26
20:56:19.787 -> D: pin_enc_sw set to 27
20:56:19.787 -> D: pin_tft_cs set to 15
20:56:19.787 -> D: pin_tft_dc set to 2
20:56:19.787 -> D: pin_tft_scl set to -1
20:56:19.787 -> D: pin_tft_sda set to -1
20:56:19.787 -> D: pin_tft_bl set to -1
20:56:19.787 -> D: pin_tft_blx set to -1
20:56:19.787 -> D: Pin 15 is already reserved!
20:56:19.825 -> D: pin_sd_cs set to 15
20:56:19.825 -> D: pin_ch376_cs set to -1
20:56:19.825 -> D: pin_ch376_int set to -1
20:56:19.825 -> D: pin_vs_cs set to 5
20:56:19.825 -> D: pin_vs_dcs set to 16
20:56:19.825 -> D: pin_vs_dreq set to 4
20:56:19.825 -> D: pin_shutdown set to -1
20:56:19.825 -> D: pin_shutdownx set to -1
20:56:19.825 -> D: pin_spi_sck set to 18
20:56:19.825 -> D: pin_spi_miso set to 19
20:56:19.825 -> D: pin_spi_mosi set to 23
20:56:19.825 -> D: GPIO0 is HIGH
20:56:19.825 -> D: GPIO2 is HIGH
20:56:19.862 -> D: GPIO4 is HIGH
20:56:19.862 -> D: GPIO5 is HIGH
20:56:19.862 -> D: GPIO12 is HIGH
20:56:19.862 -> D: GPIO13 is HIGH
20:56:19.899 -> D: GPIO14 is HIGH
20:56:19.899 -> D: GPIO15 is HIGH
20:56:19.899 -> D: GPIO16 is HIGH
20:56:19.937 -> D: GPIO17 is HIGH
20:56:19.937 -> D: GPIO18 is HIGH
20:56:19.937 -> D: GPIO19 is HIGH
20:56:19.937 -> D: GPIO21 is HIGH
20:56:19.974 -> D: GPIO22 is HIGH
20:56:19.974 -> D: GPIO23 is HIGH
20:56:19.974 -> D: GPIO25 is LOW, probably no PULL-UP
20:56:19.974 -> D: GPIO26 is LOW, probably no PULL-UP
20:56:20.009 -> D: GPIO27 is HIGH
20:56:20.009 -> D: GPIO32 is HIGH
20:56:20.009 -> D: GPIO33 is HIGH
20:56:20.047 -> D: GPIO34 is LOW, probably no PULL-UP
20:56:20.047 -> D: GPIO35 is HIGH
20:56:20.047 -> D: GPIO39 is LOW, probably no PULL-UP
20:56:20.047 -> D: gpio_00 will execute uppreset = 1
20:56:20.047 -> D: gpio_12 will execute upvolume = 2
20:56:20.047 -> D: gpio_13 will execute downvolume = 2
20:56:20.047 -> D: Enable pin 35 for IR
20:56:20.084 -> D: Start display
[... 97 lines cut ...]
20:56:47.194 -> D: Headerline: Cache-Control: no-cache, no-store
20:56:47.231 -> D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
20:56:47.231 -> D: Headerline: Connection: Close
20:56:47.231 -> D: Headerline: Access-Control-Allow-Origin: *
20:56:47.231 -> D: Headerline: Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type, Icy-MetaData
20:56:47.231 -> D: l-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type, Icy-MetaData seen.
20:56:47.231 -> D: Headerline: Access-Control-Allow-Methods: GET, OPTIONS, SOURCE, PUT, HEAD, STATS
20:56:47.264 -> D: Headerline: icy-metaint:16000
20:56:47.264 -> D: Switch to DATA, bitrate is 192, metaint is 16000
20:56:47.264 -> D: Duration mp3loop 80
20:56:47.264 -> D: Metadata block 64 bytes
20:56:47.301 -> D: Streamtitle found, 52 bytes
20:56:47.301 -> D: StreamTitle='NPO Radio 1 - NOS Langs de Lijn - NOS';
20:56:53.075 -> D: Command client available
20:56:53.075 -> D: handlehttp started
20:56:53.075 -> D: Get command is: status
20:56:53.075 -> D: Send reply for status
20:56:53.075 -> D: Command: status with parameter 0
20:56:54.202 -> D: Command client available
20:56:54.202 -> D: handlehttp started
20:56:54.202 -> D: Filename is: index.html
20:56:54.202 -> D: Start file reply for index.html
20:56:54.239 -> D: FileRequest received index.html
20:56:54.239 -> D: Length of page is 8474
20:56:54.239 -> D: Response send

Edzelf, 2022-04-23:
You may try version 2.

Fabri54, 2022-04-23:
I believe I'm using it. When is dated version 2?

Edzelf, 2022-04-24:
V2 is at https://github.com/Edzelf/ESP32Radio-V2.

Fabri54, 2022-04-24:
Thank you ,but the V2 seems not to support  the SD nor the USB. It's not a critical feature but, as I'm planning to assemble a final version in a wooden enclosure I was planning to have a slot for SD. or an USB plug.
Well, in any case it's a great project. Thank you again

Edzelf, 2022-04-24:
The choice is yours, but the V2 version does support SD, although the number of tracks is limited.

Fabri54, 2022-04-24:
Ah... I read the introductory page and stopped there after reading "SD cards and USB not supported yet". I think I'll give a try after having polished the V1 implementation for "production"....(my wife like this Internet Radio)
The amount of coding and its elegance are impressive even for a beginner like me.

synack-dev, 2022-11-08:
Looks like you can play the song by prepending localhost/ to the filename.mp3
I'll take a look why the drop down is empty and why the "random" button misses the /