Edzelf/ESP32-Radio #506: AAC VBR stream quality drops after a few seconds

FlintEastwood, 2022-05-02:
Hello,
I have this problem with this station:
`frontend.streamonkey.net/radioc-live`

When the stream starts, everything sounds fine, but after 15-20 seconds the sound quality drops and stays that way. It sounds like a really low bitrate.
When I switch to another station and then back, it sounds fine again for 15-20 seconds and then quality drops again.

The serial monitor doesn't show anything special:

> 22:04:45.440 -> D: Command: downpreset with parameter 1
> 22:04:45.487 -> D: Preset is now 0
> 22:04:45.487 -> D: STOP requested
> 22:04:45.487 -> D: Stopping client
> 22:04:45.956 -> D: New preset/file requested (0/0) from frontend.streamonkey.net/radioc-live
> 22:04:46.003 -> D: New station request
> 22:04:46.003 -> D: Connect to new host frontend.streamonkey.net/radioc-live
> 22:04:46.003 -> D: Connect to frontend.streamonkey.net on port 80, extension /radioc-live
> 22:04:46.050 -> D: Connected to server
> 22:04:46.144 -> D: Song stopped correctly after 20 msec
> 22:04:47.082 -> D: Switch to HEADER
> 22:04:47.082 -> D: Headerline: Access-Control-Allow-Origin: *
> 22:04:47.082 -> D: Headerline: Cache-Control: no-cache
> 22:04:47.082 -> D: Headerline: Content-Type: text/html; charset=utf-8
> 22:04:47.082 -> D: text/html; charset=utf-8 seen.
> 22:04:47.082 -> D: Headerline: Location: http://edge60.streamonkey.net/radioc-live
> 22:04:47.082 -> D: Headerline: Server: streaMonkey streaming Server Loadbalancer Native
> 22:04:47.082 -> D: Headerline: Content-Length: 64
> 22:04:47.082 -> D: Switch to DATA, bitrate is 0, metaint is 0
> 22:04:47.082 -> D: New station request
> 22:04:47.082 -> D: Connect to new host edge60.streamonkey.net/radioc-live
> 22:04:47.129 -> D: Connect to edge60.streamonkey.net on port 80, extension /radioc-live
> 22:04:47.176 -> D: Connected to server
> 22:04:48.160 -> D: Switch to HEADER
> 22:04:48.160 -> D: Headerline: Access-Control-Allow-Credentials: true
> 22:04:48.207 -> D: Headerline: Access-Control-Allow-Origin: *
> 22:04:48.207 -> D: Headerline: Access-Control-Expose-Headers: SessionId, icy-name, icy-description, icy-pub, icy-genre, icy-url, X-Websocket-Url
> 22:04:48.207 -> D: Headerline: Cache-Control: no-cache
> 22:04:48.207 -> D: Headerline: Connection: close
> 22:04:48.207 -> D: Headerline: Content-Type: audio/aac
> 22:04:48.207 -> D: audio/aac seen.
> 22:04:48.207 -> D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
> 22:04:48.207 -> D: Headerline: Pragma: no-cache
> 22:04:48.207 -> D: Headerline: Server: streaMonkey streaming Server Native
> 22:04:48.207 -> D: Headerline: Sessionid: 6270395e6324be439287f870
> 22:04:48.241 -> D: Headerline: Set-Cookie: streamingsession=6270395e6324be439287f870; Path=/; Domain=streamonkey.net; Max-Age=9999999999; SameSite=Lax
> 22:04:48.241 -> D: Headerline: Set-Cookie: sm_websocket_url=wss://edge60.streamonkey.net:443/wstitleupdate/6270395e6324be439287f870; Path=/; Domain=streamonkey.net; Max
> 22:04:48.241 -> D: Headerline: X-Websocket-Url: wss://edge60.streamonkey.net:443/wstitleupdate/6270395e6324be439287f870
> 22:04:48.287 -> D: Headerline: icy-audio-info: channels=2;samplerate=44100
> 22:04:48.287 -> D: Headerline: icy-description: Radio C Luxembourg
> 22:04:48.287 -> D: Headerline: icy-genre: Rock
> 22:04:48.287 -> D: Headerline: icy-metaint: 8192
> 22:04:48.287 -> D: Headerline: icy-name: Radio C Luxembourg
> 22:04:48.287 -> D: Headerline: icy-pub: 1
> 22:04:48.287 -> D: Headerline: icy-url: https://radioc.eu
> 22:04:48.287 -> D: Switch to DATA, bitrate is 0, metaint is 8192
> 22:04:48.287 -> D: Metadata block 32 bytes
> 22:04:48.287 -> D: Streamtitle found, 22 bytes
> 22:04:48.287 -> D: StreamTitle='Radio C';
> 22:04:48.475 -> D: Metadata block 32 bytes
> 22:04:48.475 -> D: Streamtitle found, 22 bytes
> 22:04:48.475 -> D: StreamTitle='Radio C';
> 22:04:49.342 -> D: Metadata block 32 bytes
> 22:04:49.342 -> D: Streamtitle found, 22 bytes
> 22:04:49.342 -> D: StreamTitle='Radio C';
> 22:04:50.072 -> D: Metadata block 32 bytes
> 22:04:50.072 -> D: Streamtitle found, 22 bytes
> 22:04:50.120 -> D: StreamTitle='Radio C';

Edzelf, 2022-05-02:
No problem here.  But I used the V2 version of the radio.
The test-button shows:
`Free memory is 113792, chunks in queue 395, bitrate 58 kbps`

Rainer-G, 2022-05-04:
I tested the station frontend.streamonkey.net/radioc-live with Winamp and saw the bitrate varying from less than 50kbps to more than 150 kbps. I could not hear any loss in quality.With V1, quality is as in Winamp but the diplay of the bitrate does not follow the variable bitrate, looks that the data for display is not refreshed evry second as Winamp does. Rainer Gesendet von Mail für Windows Von: Ed SmallenburgGesendet: Dienstag, 3. Mai 2022 00:18An: Edzelf/ESP32-RadioCc: SubscribedBetreff: Re: [Edzelf/ESP32-Radio] AAC VBR stream quality drops after a few seconds (Issue #506) No problem here. But I used the V2 version of the radio.The test-button shows:Free memory is 113792, chunks in queue 395, bitrate 58 kbps

Edzelf, 2022-05-04:
Unreadable.

Rainer-G, 2022-05-04:
sorry, Ed, copy & paste put in some HTML coding, this should be better:

I tested the station frontend.streamonkey.net/radioc-live with Winamp and saw the bitrate varying from less than 50kbps to more than 150 kbps. I could not hear any loss in quality.
With V1, quality is as in Winamp but the diplay of the bitrate does not follow the variable bitrate, looks that the data for display is not refreshed evry second as Winamp does.
 
Rainer

Edzelf, 2022-05-05:
Just tested with V2 and I2S output: no problem.
Just tested with V2 and VS1053 output: no problem.
What do you mean by the display of the bitrate?  This is only displayed in the web interface for test purposes.  It is the bitrate averaged over 10 seconds.  Not intended to give a precise real-time measurement.

Rainer-G, 2022-05-05:
Hello Ed, I meant the bitrate displayed using the test button on the web interface. I do not see this as an issue and there is no need for a change. Maybe an indicator that the data comes from a VBR station can avoid such irritations?I remember a similar issue (#476?) was reported some time ago which also came from a VBR station. Rainer Gesendet von Mail für Windows Von: Ed SmallenburgGesendet: Donnerstag, 5. Mai 2022 08:26An: Edzelf/ESP32-RadioCc: Rainer-G; CommentBetreff: Re: [Edzelf/ESP32-Radio] AAC VBR stream quality drops after a few seconds (Issue #506) Just tested with V2 and I2S output: no problem.Just tested with V2 and VS1053 output: no problem.What do you mean by the display of the bitrate? This is only displayed in the web interface for test purposes. It is the bitrate averaged over 10 seconds. Not intended to give a precise real-time measurement.

FlintEastwood, 2022-05-05:
Hello Ed,
Thank you for your reply. 
I needed two evenings to get Platformio running. 
Yesterday I loaded V2 into my Esp32.
But the problem with "Radio C" is still the same.
I'm using a AZ Delivery Esp32 Dev Kit v4 (changed board setting in the Platformio.ini) and one of those green square Vs1053 pcb's. But that should work, shouldn't it?
There is a Test button in the Web interface? I need to check this tonight.
Greetings,  Stefan

Edzelf, 2022-05-05:
Why don't you install PlatformIO in 5 minutes?  That saves some time ;)
The board usually does not matter.  Be sure there is a real VS1053 chip on it,  There are lots of boards sold with the wrong chip.

FlintEastwood, 2022-05-07:
Haha yes, install was fast, but for me getting used to, took a bit more time. ;)
The chip on my green PCB is a real VS1053B .

I modified your code just for testing (should I fork this?). I made a similar function as the Timer10sec function called Timer2sec, but executed every two seconds to calculate the bitrate and print it via "mbitreq=true" to the serial monitor.
Now you can see that the bitrate drops after a few seconds and that is what I hear.
The same chunk recorded with streamwriter on PC has an average bitrate of 87kbps. Much higher than the ESP32radio reports.

Is it possible, that the streamonkey link has multiple streams with different bitrates and the ESP32radio switches (or is forced) to a lower bitrate? I really have no knowledge about such things.

`D: IR code 50AF received. Will execute downpreset = 1
D: updateNr 1 <= 7 to -1, relative is 1
D: updateNr result is 0
D: nextPreset is 0
D: Preset is now 0
D: Radiofuncs cmd is 1
D: Stopping client
D: Connect to host frontend.streamonkey.net/radioc-live
D: send GET command
D: Switch to HEADER
D: Headerline: Access-Control-Allow-Origin: *
D: Headerline: Cache-Control: no-cache
D: Headerline: Content-Type: text/html; charset=utf-8
D: Headerline: Location: http://edge63.streamonkey.net/radioc-live  
D: Headerline: Server: streaMonkey streaming Server Loadbalancer Native
D: Headerline: Content-Length: 64 
D: Redirect
D: Switch to HEADER
D: Radiofuncs cmd is 1
D: Connect to host edge63.streamonkey.net/radioc-live
D: send GET command
D: Song stopped correctly after 20 msec
D: Switch to HEADER
D: Headerline: Access-Control-Allow-Credentials: true
D: Headerline: Access-Control-Allow-Origin: *
D: Headerline: Access-Control-Expose-Headers: SessionId, icy-name, icy-description, icy-pub, icy-genre, icy-url, X-Websocket-Url
D: Headerline: Cache-Control: no-cache
D: Headerline: Connection: close  
D: Headerline: Content-Type: audio/aac
D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Headerline: Pragma: no-cache   
D: Headerline: Server: streaMonkey streaming Server Native
D: Headerline: Sessionid: 62762e18e0654511db2c3b5e
D: Headerline: Set-Cookie: streamingsession=62762e18e0654511db2c3b5e; Path=/; Domain=streamonkey.net; Max-Age=9999999999; SameSite=Lax  
D: Headerline: Set-Cookie: sm_websocket_url=wss://edge63.streamonkey.net:443/wstitleupdate/62762e18e0654511db2c3b5e; Path=/; Domain=streamonkey.net; Max-Age=9999999999; SameSite=Lax
D: Headerline: X-Websocket-Url: wss://edge63.streamonkey.net:443/wstitleupdate/62762e18e0654511db2c3b5e
D: Headerline: icy-audio-info: channels=2;samplerate=44100
D: Headerline: icy-description: Radio C Luxembourg
D: Headerline: icy-genre: Rock    
D: Headerline: icy-metaint: 8192
D: Headerline: icy-name: Radio C Luxembourg
D: Headerline: icy-pub: 1
D: Headerline: icy-url: https://radioc.eu
D: Switch to DATA, bitrate is 0, metaint is 8192
D: Streamtitle found, 40 bytes
D: StreamTitle='Wir sind Helden - Denkmal';
D: Streamtitle found, 40 bytes
D: StreamTitle='Wir sind Helden - Denkmal';
D: Free memory is 69620, chunks in queue 400, bitrate 4294611 kbps  
D: Streamtitle found, 40 bytes
D: StreamTitle='Wir sind Helden - Denkmal';
D: Streamtitle found, 40 bytes
D: StreamTitle='Wir sind Helden - Denkmal';
D: Free memory is 67572, chunks in queue 400, bitrate 69 kbps       
D: Streamtitle found, 40 bytes
D: StreamTitle='Wir sind Helden - Denkmal';
D: Free memory is 67572, chunks in queue 400, bitrate 71 kbps       
D: Free memory is 67572, chunks in queue 400, bitrate 64 kbps       
D: Free memory is 69620, chunks in queue 400, bitrate 72 kbps       
D: Free memory is 67572, chunks in queue 400, bitrate 66 kbps       
D: Free memory is 65524, chunks in queue 400, bitrate 79 kbps       
D: Free memory is 69620, chunks in queue 400, bitrate 69 kbps       
D: Free memory is 67572, chunks in queue 400, bitrate 53 kbps       <-- Here drops the quality!
D: Free memory is 67572, chunks in queue 400, bitrate 50 kbps       
D: Free memory is 69620, chunks in queue 400, bitrate 49 kbps       
D: Free memory is 67572, chunks in queue 400, bitrate 49 kbps       
D: Free memory is 69620, chunks in queue 400, bitrate 48 kbps
D: Free memory is 67572, chunks in queue 400, bitrate 49 kbps
D: Free memory is 67572, chunks in queue 400, bitrate 51 kbps
D: Free memory is 69620, chunks in queue 400, bitrate 46 kbps
D: Free memory is 67572, chunks in queue 400, bitrate 48 kbps
D: Free memory is 67572, chunks in queue 400, bitrate 49 kbps`

Greetings, Stefan