Edzelf/ESP32-Radio #508: Complete tragedy - no OLED, not user friendly install etc

Sentinel8000, 2022-05-17:
I was realy happy after i have found this project because this looked realy a complete full circle project with lot display support. But lookslike this project working almost if all stars stay is good position. My first problem was, not realy clear the install steps, i need run first the radio init, or after? First i try simple complie and upload the code itself with arduino ide, but first comes just problem, lookslike AP 192.168.4.1 ip not compatible with my network. Step by steps instruction realy missing, this comedy video on youtube what was linked funny but not helpfully:)

After i try split my problems to steps, ok first i will be happy to see my 1106 oled display to works or a wifi. Not realy clear after i have uncomment my oled header after i need do anything or no. After i have found this ,,preference" ino and i try add:

  preferences.putString ( "pin_tft_scl",  "22                                     # GPIO Pin number for TFT CS" ) ;
  preferences.putString ( "pin_tft_sda",  "21                                      # GPIO Pin number for TFT DC" ) ;

but still no picture. 

However the code is almost well documented, not realy easy understand where i can change this pins directly.

After i have uploaded this prefrence ini, i was happy, because network working, but AP method (what im sure can be a must have/good feature)  with the fixed IP not. Code is not ready change this ip into a custom address. I dont know possible to print into a serial monitor the received ip, because after conenction, i was blind what ip received my esp32:

07:06:37.690 -> D: Connect to WiFi
07:06:42.741 -> D: Start server for commands
07:06:42.741 -> D: Network found. Starting mqtt and OTA

last time i try first upload preferences ini and after the radio code. But currently see weird anomaly in prefrences:

TKEY_00 = <someting>
TKEY_01 = <someting>
TKEY_02 = <someting>
TKEY_03 = <someting>
TKEY_04 = <someting>
TKEY_05 = <someting>
TKEY_06 = <someting>
TKEY_07 = <someting>
TKEY_08 = <someting>
TKEY_09 = <someting>
TKEY_10 = <someting>
TKEY_11 = <someting>
TKEY_12 = <someting>
TKEY_13 = <someting>
TKEY_14 = <someting>
TKEY_15 = <someting>
TKEY_16 = <someting>
TKEY_17 = <someting>
TKEY_18 = <someting>
TKEY_19 = <someting>
TKEY_20 = <someting>
TKEY_21 = <someting>
TKEY_22 = <someting>
TKEY_23 = <someting>
TKEY_24 = <someting>
TKEY_25 = <someting>
TKEY_26 = <someting>
TKEY_27 = <someting>
TKEY_28 = <someting>
TKEY_29 = <someting>
#
preset = -1
#
toneha = 0
tonehf = 0
tonela = 0
tonelf = 0
#
volume = 0

And also still no display. What i need do to check settings for OLED 1106? where can i find the pin settings in code? What is the install/upload steps? why i see this weird data in config after upload sketch? 

thank you

Edzelf, 2022-05-17:
Please switch to PlatformIO and to version 2 of the radio.
There is no need to fill the preferences with the init sketch.  Just edit config.h
And read the manual.

Sentinel8000, 2022-05-17:
I have never used platformIO, maybe i will try.  config.h, where?

I think the main problem is the complexity of this project with support lot things, i read lot comments before i place this issue here, lot people fighting to bring this project life:) This can be more user friendly if will be just a minimalistic edition with Serial monitor display and web server (this can be a good thing because can change settings without upload a code) and if somebody want add itself any display commands.

I try again..somehow..., but maybe i need try find a more minimalistic project and try customize.

However thank you for your work i see the invested workhours inside.