Edzelf/ESP32-Radio #509: Define GPIO pins for VS1053

dansity, 2022-05-31:
Hello! Thank you for your work. I have read the manual but stuck with the pin setup.
The main.ccp file has an "example" wiring and mentiones that I will be able to change these once I'm up and running. Now the firmware is uploaded but the config has only a handful of pins defined.
For the VS1053 there are 3 pins defined while it has 7
pin_vs_cs = 5
pin_vs_dcs = 32
pin_vs_dreq = 4
I'm missing MOSI,MISO,SCK,XRST
Also I need the pin setup for the screen I will use (OLED, I need SCK and SDA pins).
How to define them?

Thank you

Edzelf, 2022-05-31:
Some of the pins have a default assignment, see table progpin[] in main.cpp.
For example pins 18,19,21,22,23:
  { 18, false, false,  "", false },                      // Default for SPI CLK
  { 19, false, false,  "", false },                      // Default for SPI MISO
  { 21, false, false,  "", false },                      // Also Wire SDA
  { 22, false, false,  "", false },                      // Also Wire SCL
  { 23, false, false,  "", false },                      // Default for SPI MOSI

In the config page, you only have to specify pins that differ from the default.

dansity, 2022-06-07:
Really thank you for your reply. As my upper table shows I have used the same pins (even if I have defined them again) but I'm not able to figure out what is the issue. I have the ESP showing the radio playing but I have no sound and I have no image on the oled. I have checked wiring and continuity at least 10 times now, everything is by the book.
If you have a few tips what to check it would be awsome