Edzelf/ESP32-Radio #511: main.cpp does not compile

fanfanlatulipe26, 2022-11-14:
main.cpp does not compile on Arduino IDE 1.8.19
In main.cpp, the line 151   is : 
` triplecl// 23-03-2020, ES: Allow playlists on SD card.`

Seems that the "triplecl" is a typo comming from ?????

Edzelf, 2022-11-14:
Fixed.  But please switch to version 2 of the radio.

fanfanlatulipe26, 2022-11-14:
Will do

fanfanlatulipe26, 2022-11-15:
Done.  
I use the box of an old web radio  Phoenix Com One  
[image]  
and made some changes in the software to also use the preset buttons of the box. 
I ported also these changes in the version 2 and it runs like a charm. 
Thank you for your work.

Rillez, 2022-12-23:
looks really nice