Edzelf/ESP32-Radio #512: Few qeustions before building

HaViGit, 2023-01-07:
Before I start building this nice project a few questions:

1. Do I need Platformio?
2. Which Nextion display can I use?
3. I will only use Bluetooth, which version (V1 or V2) do you recommend and the VS1053 or the PCM5102A board?

I hope you have time to advise, thanks in advance!

Edzelf, 2023-01-07:
1. Platformio is the best option.  For the Ardino IDE you need to modify some filenames and organize the used libraries.
2. The easiest way is to use the same Nextion as I did.  For other types you have to adjust te coding.
3. You cannot use Bluetooth and WiFi at the same time in the ESP32.  So you need a Bluetooth transmitter.  The VS5102A support more codecs than the software driving the PCM5102A.

Please use the V2 version of the radio software!

HaViGit, 2023-01-07:
Thanks for the quick response Ed! 

Which Nextion display did you use? It is indeed my intention to use a BT audio transmitter, that is no problem. Is the PCM5102A sufficient to simply reproduce audio in your project? Or do I need the extra codecs?

Edzelf, 2023-01-07:
The type of Nextion should be in the documentation.  There is software codes for MP3 and AAC, so most stations will work.

closed 2023-01-07 by HaViGit