Edzelf/ESP32-Radio #515: Adding LCD screen created choppy audio

garbo5005, 2023-01-21:
I have started on this great project and did get it working, The problem I am having and hoping someone can help me resolve is that when I have the esp32 connected just to the VS1053 ( which is actually a VS1003 ) I have great audio and no problems with the web interface. When I add the screen ( I have tried two) I get choppy audio. It seems like adding the screen slows down processing or something. Has anyone experience this problem ?

Edzelf, 2023-01-21:
SPI, I2C?
Please try Version 2 of the radio.

aly-fly, 2023-01-21:
Choppy audio can also be caused by bad wiring. SPI bus carries high frequency signals that are time sensitive and longer / messy wires will cause all sorts of problems. Especially if you have two SPI devices wired to the same bus. Best length of total SPI wires is < 5 cm.

garbo5005, 2023-01-21:
Thank you both for the reply. I am using V2 but I am also using 200mm cables with my bread board so that is the first place I will try.

garbo5005, 2023-01-21:
Well the cleaning up the wiring helped! I actually just committed to soldering to a board since this has been a proven and well working project for others thank you for the help. 
[image]

aly-fly, 2023-01-22:
Good.
My setup is like this and works fine at 3 MHz SPI frequency.

[image]