Edzelf/ESP32-Radio #516: Newbie Help

prairietech, 2023-03-12:
I found an excellent review by Andreas Spiess and need some help trying to compile this on Arduino IDE. Adding Libraries hasn't been an issue for me in the past but adding nvr.h is my first brickwall. Maybe some one on one email support from a user?
Thank you! rlnou812@gmail.com

Edzelf, 2023-03-12:
I don't know what "nvr.h" is.
If you mean "nvs.h", that part of the standard Espressif library (non-volatile storage).

It is better to use the version 2 of the Radio, see [https://github.com/Edzelf/ESP32Radio-V2](https://github.com/Edzelf/ESP32Radio-V2)

prairietech, 2023-03-12:
Oh wow! I was using the old version! OK, I'll dive into version 2 and see how it goes.
Thanks for the quick reply!

prairietech, 2023-03-12:
Making progress, and nvs.h is what I intended, fat fingers.
But I'm still not finding nvs.h or the standard Espressif library using the Manage Library search.

Edzelf, 2023-03-12:
It should be internal, no library needed, like math.h or studio.h or stdlib.h.

prairietech, 2023-03-12:
During verify, I get an error, nvs.h: No such file

Edzelf, 2023-03-12:
I will check it tomorrow, but you can avoid these problems by using Platformio.  You won't regret it!

prairietech, 2023-03-12:
Thank you!  I did find esp-idf-master that contains nvs.h but the Arduino IDE throws an error stating no .h files exist.
I did look at Platformio, but it looks pretty intense for someone of my low level.

Edzelf, 2023-03-13:
Are you sure you have selected "esp32-devkit" for your board?
Devices like ArduinoUno do not have NVS.

Edzelf, 2023-03-13:
This example compiles without errors.  Please try to reproduce it.
Note the "ESP32 Dev Module" at the top of the screen print.

[image]

I also tried the 1.8.13 version of the Arduino IDE; no problems found.

prairietech, 2023-03-14:
Making slow progress. Started Arduino IDE and tried a compile again.  This time the verify went past nvs.h but is now stopped on ESPAsyncWebserver.h with an error claiming no such file.  However checking the Library Manager, the name is ESPAsyncWebSrv and version 1.2.6 is installed.  You specify version 1.2.0 but apparently that's not available any more.

Edzelf, 2023-03-14:
Check the right spelling.  It should be:
#include <ESPAsyncWebServer.h>
With a capital "S".

aly-fly, 2023-03-14:
I was also trying to get Radio V2 working in Arduino IDE, but gave up very quickly.

My strong recommendation is to install Platformio. Guides are very easy to follow and you get everything set up way faster than hunting for typos like this above.

prairietech, 2023-03-15:
So I installed Platformio and imported ESP32Radio. Setup for the correct board and did a Build.  Now to sort through the 16 errors found in the .ino and 49 errors in helixfunc.h and other various errors and problems in different files.
It's hard to teach this old dog new tricks but I'm trying!!
Thanks for the support and suggestions!  Without it I would have given up.

aly-fly, 2023-03-15:
Delete the project you were working on and download fresh copy of Radio V2 from github repo. Delete the .ino file if you have any. Platformio does not support .ino files, these are for Arduino only.

prairietech, 2023-03-15:
aly-fly.....thanks!  I remember seeing that detail but forgot about it. Deleted everything and started over fresh.
Watched some tutorials on Platformio and have made major progress compared to using the Arduino IDE!
However there are 48 errors in helixfunc.h but since I'm using a VS1053, do those errors need to be corrected?

It appears most of those errors are identifier xxxxx is undefined at various line numbers.  If the errors need to be corrected, I'm not sure how to correct most of them?

Edzelf, 2023-03-16:
Are you using Radio V2 now?
Show me your config.h

prairietech, 2023-03-17:
Probably something silly I did or didn't do.

[{
	"resource": "/C:/Users/User/Downloads/ESP32Radio-V2-main/ESP32Radio-V2-main/include/helixfuncs.h",
	"owner": "C/C++",
	"code": "20",
	"severity": 8,
	"message": "identifier \"String\" is undefined",
	"source": "C/C++",
	"startLineNumber": 11,
	"startColumn": 8,
	"endLineNumber": 11,
	"endColumn": 14
},{
	"resource": "/C:/Users/User/Downloads/ESP32Radio-V2-main/ESP32Radio-V2-main/include/helixfuncs.h",
	"owner": "C/C++",
	"code": "20",
	"severity": 8,
	"message": "identifier \"int16_t\" is undefined",
	"source": "C/C++",
	"startLineNumber": 13,
	"startColumn": 8,
	"endLineNumber": 13,
	"endColumn": 15
},{
	"resource": "/C:/Users/User/Downloads/ESP32Radio-V2-main/ESP32Radio-V2-main/include/helixfuncs.h",
	"owner": "C/C++",
	"code": "20",
	"severity": 8,
	"message": "identifier \"uint8_t\" is undefined",
	"source": "C/C++",
	"startLineNumber": 15,
	"startColumn": 8,
	"endLineNumber": 15,
	"endColumn": 15
},{
	"resource": "/C:/Users/User/Downloads/ESP32Radio-V2-main/ESP32Radio-V2-main/include/helixfuncs.h",
	"owner": "C/C++",
	"code": "20",
	"severity": 8,
	"message": "identifier \"uint8_t\" is undefined",
	"source": "C/C++",
	"startLineNumber": 16,
	"startColumn": 8,
	"endLineNumber": 16,
	"endColumn": 15
},{
	"resource": "/C:/Users/User/Downloads/ESP32Radio-V2-main/ESP32Radio-V2-main/include/helixfuncs.h",
	"owner": "C/C++",
	"code": "20",
	"severity": 8,
	"message": "identifier \"int16_t\" is undefined",
	"source": "C/C++",
	"startLineNumber": 19,
	"startColumn": 8,
	"endLineNumber": 19,
	"endColumn": 15
},{
	"resource": "/C:/Users/User/Downloads/ESP32Radio-V2-main/ESP32Radio-V2-main/include/helixfuncs.h",
	"owner": "C/C++",
	"code": "20",
	"severity": 8,
	"message": "identifier \"int16_t\" is undefined",
	"source": "C/C++",
	"startLineNumber": 27,
	"startColumn": 25,
	"endLineNumber": 27,
	"endColumn": 32
},{
	"resource": "/C:/Users/User/Downloads/ESP32Radio-V2-main/ESP32Radio-V2-main/include/helixfuncs.h",
[... 436 lines cut ...]
	"endLineNumber": 189,
	"endColumn": 13
},{
	"resource": "/C:/Users/User/Downloads/ESP32Radio-V2-main/ESP32Radio-V2-main/include/helixfuncs.h",
	"owner": "C/C++",
	"code": "20",
	"severity": 8,
	"message": "identifier \"i2s_write\" is undefined",
	"source": "C/C++",
	"startLineNumber": 210,
	"startColumn": 5,
	"endLineNumber": 210,
	"endColumn": 14
},{
	"resource": "/C:/Users/User/Downloads/ESP32Radio-V2-main/ESP32Radio-V2-main/include/helixfuncs.h",
	"owner": "C/C++",
	"code": "20",
	"severity": 8,
	"message": "identifier \"portMAX_DELAY\" is undefined",
	"source": "C/C++",
	"startLineNumber": 211,
	"startColumn": 17,
	"endLineNumber": 211,
	"endColumn": 30
}]

Edzelf, 2023-03-17:
Looks like the line:
`#include` `<Arduino.h>`
Is missing at the beginning of main.cpp

prairietech, 2023-03-17:
Arduino.h is present:

// Define the version number, the format used is the HTTP standard.
#define VERSION     "Thu, 03 Nov 2022 13:00:00 GMT"
//
#include <Arduino.h>                                      // Standard include for Platformio Arduino projects
#include <WiFi.h>
#include "config.h"                                       // Specify display type, decoder type
#include <nvs.h>                                          // Access to NVS
#include <PubSubClient.h>                                 // MTTQ access
#ifdef ETHERNET
  // Definitions for Ethernet controller
  #include <ETH.h>
  #define ETH_CLK_MODE    ETH_CLOCK_GPIO0_IN              // External clock from crystal oscillator
  #define ETH_TYPE        ETH_PHY_LAN8720                 // Type of controller
  #define ETH_ADDR        1                               // I2C address of Ethernet PHY
#else
  #include <WiFiMulti.h>                                  // Handle multiple WiFi networks
  WiFiMulti wifiMulti ;                                   // Object for WiFiMulti

prairietech, 2023-03-18:
I "think" I'm making slow progress?
But when uploading firmware and data, in platformia.ini I have set monitor_port and upload_port to COM8. 
However when in terminal and issue pio run -t upload, I see it trying to connect to COM3 followed by an upload error 2.

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
monitor_port = COM8
upload_port = COM8

prairietech, 2023-03-18:
I'm making good progress but can't seem to connect to my internet wifi AP. 
My AP is defined in config.h under FIXEDWIFI along with it's password (which is all numeric, I have other esp32s running successfully with this all numeric password if that matters)

From the serial monitor, I show 1 available network. My wifi AP is then listed with a -27dbm signal level. A few lines later, connect to network, a pause, then wifi failed. 
The internal AP fires up and I can connect to ESP32-Radio just fine. From the web page Config, my wifi AP is listed as available. I have tried entering my wifi AP info at wifi_01 on the Config web page without success.

Ideas / suggestions??

NickBramley, 2023-03-18:
Hi,

                I have been having similar problems to you. What I have done is start afresh with a new install of the software from github. Open it in PlatformIO as described. Do not edit anything! I did run ES32_radio_init.ino from the Arduino environment, then main.cpp from Pio. When the radio cannot connect to a wifi, it will make the AP ESP32-Radio (with the hyphen) and you can then add your SSID and pw as described. The AP password also needs the hyphen.

Good luck.

 

Nick

NickBramley, 2023-03-18:
…I also had to change the gpio default pin to gpio 16 for the VS1053 XDCS.

 

Nick

Edzelf, 2023-03-18:
Sometimes, a complete powerdown helps.
Sometimes, a complete flash erase, followed by a fresh upload helps.
Note that ESP32 cannot connect to 5 GHz accesspoint.

Edzelf, 2023-03-18:
A good ower supply is also very important.

prairietech, 2023-03-18:
My first error was using a / between the ssid and the password instead of a hyphen - since the example was showing a /
But how many places does this info need to be entered?
#define FIXEDWIFI in config.h and wifi_01 from the web page?

I've changed from / to a - between ssid and password without success so far. 
Running a 5.25 volt power supply capable of 2 amps so that should be good.

Edzelf, 2023-03-18:
A slash is expected.

NickBramley, 2023-03-18:
Hi Ed,

                I am just blown away by your software skills and knowledge and now relieved to get it working!

                Are there any pcb’s available for the radio? I can lay out a new version but buying one is a bit easier!!

 

Nick

prairietech, 2023-03-18:
OK, back to a slash.
But does ssid and pw need to be in both FIXEDWIFI and wifi_0 of the webpage Config page?

NickBramley, 2023-03-18:
No !

Edzelf, 2023-03-18:
Defining FIXEDWIFI is sufficient.  Use wifi_nn for extra networks.

prairietech, 2023-03-18:
Still no luck.....not sure what to try next?
I've clear flash. Started over with a fresh download.
Even pulled another new esp32 out of the box with the same results.

If I set up another AP for testing, does it have to use a password? If not, what is used for the password in the SSID/password parameter?

Edzelf, 2023-03-18:
Password is equal to SSID.

prairietech, 2023-03-18:
I set up a different internet wifi AP using a different SSID and alphanumeric password and still can't connect.
I'm so close but so far away :-(

prairietech, 2023-03-18:
Your example in the PDF shows an IP address where mine shows 1.0.0.0. 
However using my cellphone, I can connect to ESP32-Radio and browse to 192.168.4.1
Sorry for all the questions but it's frustrating to come this far then have a simple lack of wifi connection hose everything up.

Connect to network
D: WiFi Failed!  Trying to setup AP with name ESP32-Radio and password ESP32-Radio.
D: IP = 1.0.0.0
D: Start web server

NickBramley, 2023-03-18:
Sorry ED….just realised that I was emailing you rather than Prairietech!

 

Still would like to know if the pcbs are available form you or Harm Verbeek,

 

NIck

Edzelf, 2023-03-18:
I do not have PCBs.

Edzelf, 2023-03-18:
@prairietech: I cannot reproduce the 1.0.0.0 IP address.  In AP mode it should always be 192.168.4.1.

prairietech, 2023-03-18:
So is it safe to overlook this IP address when it shows up in the Serial monitor or does it represent a error that needs to be corrected?

D: WiFi Failed!  Trying to setup AP with name ESP32-Radio and password ESP32-Radio.
D: IP = 1.0.0.0
D: Start web server

Edzelf, 2023-03-18:
If you can connect to it, it would be strange.
There is definitively something wrong, but I do not have the solution.
You may try an ESP32 without any wiring.  And if that work, start wiring step by step.
Maybe you can share the logging.

prairietech, 2023-03-18:
I tried your suggestion of another fresh esp32, just by itself sitting on the bench and no board connections...........same story, wifi failed.
There's a bigger picture I'm missing here, just not sure where to look next?

Maybe I could ship one or two of my esp32s to a successful user to be compiled and verified as working?

prairietech, 2023-03-18:
Nick, You might check here for PCBs:

https://hackaday.io/project/186284-new-life-for-an-old-radio-esp32-for-ekco-radio

prairietech, 2023-03-19:
Progress, sort of.
I've been trying various different internet wifi routers that I have around here without success.
But, I put my cellphone in tether wifi mode, and can get a connection from the esp32 to my cellphone !!
Selected a station and it appeared with artist and title on the display but no sound. Another hurdle to overcome.
I have many esp32s doing various things all connected to the same wifi router that I'm trying to connect Esp32Radio to.
Not sure why the Esp32Radio unit is so particular regarding who it connects to??
Anyone have a list of don'ts regarding connecting esp32's to wifi routers? I tried an open no-password setup and no go.