Edzelf/ESP32-Radio #517: ESP32 Won't Connect To My WiFi AP

prairietech, 2023-03-19:
The ESP32 won't connect to 3 different wifi APs I have tried but will connect to my cellphone when it's used as a wifi hotspot tether. Those wifi APs are all the same room as the ESP32.
Otherwise, I get the WiFi Failed! error.  My internet wifi AP ssid/password are entered under the FIXEDWIFI setting. Running the latest V2 download using Platformio.

When the connection fails, the ESP32 then goes into it's own AP mode which I can connect to using a laptop. I am able to submit changes from the web interface and they are accepted with the Serial Monitor confirming the action. From the web interface Config page, my Internet wifi AP is listed as available. I have also entered it under wifi_00 on the Config page, saved, then restarted only to have the connection still fail.

I can upload a simple wifi test sketch to the ESP32 and it will connect to the same Internet wifi AP entered under FIXEDWIFI.  I have other ESP32s doing other tasks around the house and wifi connection has never been an issue.

I've tried everything I can think of, so looking for ideas and suggestions of what to try next?

Edzelf, 2023-03-19:
Maybe you can replace the wifi connect routine with a one that works on your other ESP32s.  It's only a few lines.

prairietech, 2023-03-19:
As you can tell, I'm a long ways from a programmer. Could you tell me where in the code those lines are?

Edzelf, 2023-03-19:
Search for "connect".

prairietech, 2023-03-19:
1687 results in 88 files......may take awhile

Edzelf, 2023-03-19:
Start at line 3431.

prairietech, 2023-03-19:
Below is starting from line 3431 of main.cpp
Doesn't look like wifi commands?

static bool  connected = false ;                                // Connected to host or not

  if ( xQueueReceive ( radioqueue, &radiocmd, 0 ) )               // New command in queue?
  {
    dbgprint ( "Radiofuncs cmd is %d", radiocmd ) ;
    switch ( radiocmd )                                           // Yes, examine command
    {
      case QSTARTSONG:                                            // Start a new station?
        if ( sdqueue )

Edzelf, 2023-03-19:
I had the wrong version.  It is 1611:
 ` WiFi.disconnect ( true ) ;                            // After restart the router could
  WiFi.softAPdisconnect ( true ) ;                      // still keep the old connection
  vTaskDelay ( 1000 / portTICK_PERIOD_MS ) ;            // Silly things to start connection
  WiFi.mode ( WIFI_STA ) ;
  vTaskDelay ( 1000 / portTICK_PERIOD_MS ) ;
  if ( wifilist.size()  )                               // Any AP defined?
  {
    if ( wifilist.size() == 1 )                         // Just one AP defined in preferences?
    {
      winfo = wifilist[0] ;                             // Get this entry
      WiFi.begin ( winfo.ssid, winfo.passphrase ) ;     // Connect to single SSID found in wifi_xx
      dbgprint ( "Try WiFi %s", winfo.ssid ) ;          // Message to show`

Edzelf, 2023-03-20:
Channel 1 to 11 should be acceptable.  I am using 2 and 7 at home.