Edzelf/ESP32-Radio #519: SD card unable to play.

The-Valveman, 2023-04-18:
Hi i am having trouble getting the SD card MP3s to show up on the drop down list and am unable to play them.
i have tried various file/folder formats but to no avail the card is being read by the ESP32 as when i use the serial monitor the folder name and track names appear on the serial monitor window when i click the MP3 list button then it states "0 Mp3 tracks on local drive" and the ESP also states "the folder "Live in Australia" is not in the root. but this folder with its tracks are the only thing on the SD card
can you tell me what the folder/file format is needed to work as i cannot find any reference to this in the manual/documentation.

aly-fly, 2023-04-19:
Place only 1 to 5 .mp3 files in the. With very simple names, like "one.mp3". No folders. This works for me. 
Complex file names with special (especially unicode) characters make a mess.

Edzelf, 2023-04-19:
Please use the V2 version of the radio.  See https://github.com/Edzelf/ESP32Radio-V2
Note that the code is still experimental.  The memory space for the song list is tight.
Also note that not all types of SD cards will work.

The-Valveman, 2023-04-19:
@aly-fly  I have tried that but still no go. I get a message in the serial monitor "SD card mount failed!"

@Edzelf  I have looked at the version 2 but I see it is designed to be compiled in the Platformio IDE I have been using the Arduino IDE And being a fairly noobie to programming I really dont wish to take on the steep learning curve to use Platformio. so is there a way to use this in the Arduino IDE?  the current set up i have shows an error during boot up "SD card mount failed!"

Edzelf, 2023-04-20:
It is possible to use the Arduino IDE.  But installing PlatformIO is much easier.  You won't regret it.

MirceaB07, 2023-05-24:
For using SD Card to which pin of ESP I have to connect SD-CS from SD connector (TFT display)?
I didn't find it in wiring table in pdf file, but I find a definition as pin 21 in Esp32_radio_init.ino and as pin 14 in exampleconfig.ed.txt. The other 3 pins of SD connector doesn't need to be connected?  I'm using Platformio for compiling and uploading code. Thanks a lot.

Edzelf, 2023-05-24:
All input pins of the ESP32 are initialized using PULL_UP.
For SD card you need Power, ground, MISO, MOSI, CLK and CS.  If possible also "card detect".  The last 2 pins can be defined in the config page of the webinterface, like:
```
    pin_sd_cs = 14
    pin_sd_detect = 4

```
Note that using SD card is still experimental.