Edzelf/ESP32-Radio #520: How to compile in Visual Studio?

apadt, 2023-06-01:
Hi, since I'm amateur in compling in Visual Studio would be possible to made a short instruction both for version I and II of this excellent project? Thanks AJ

Edzelf, 2023-06-02:
1. In Visual Studio: install PlatformIO.
2. Download the the ESP32Radio-V2 zip file.
3. Unpack the zipfile in any directory.
4. Double-click on "ESP32-Radio.code-workspace.  PlatformIO will open automatically with the right workspace.
5. Check the com-port setting in platformio.ini for upload and monitoring.
6. Check include/config.h to configure your set-up.
7. Uload the sketch to your ESP32.
8. Start the serial monitor for debug info.

apadt, 2023-06-02:
Hartelijk dank / thanks for the info!