Edzelf/ESP32-Radio #527: Audio not working vs1053

redstoicescu, 2024-01-09:
I do not use i2s but :
[  3226][E][main.cpp:1809] reservepin(): [main] Pin 5 is already reserved!
[  3232][I][main.cpp:1911] readIOprefs(): [main] pin_i2s_bck set to 5
[  3239][E][main.cpp:1809] reservepin(): [main] Pin 32 is already reserved!
[  3245][I][main.cpp:1911] readIOprefs(): [main] pin_i2s_lck set to 32
[  3252][E][main.cpp:1809] reservepin(): [main] Pin 34 is already reserved!
[  3258][I][main.cpp:1911] readIOprefs(): [main] pin_i2s_din set to 34
[

redstoicescu, 2024-01-09:
removed but :
[ 30437][E][VS1053.cpp:150] testComm(): [VS1053] SPI error. SB:802E R1:802E R2:8700
[ 30447][E][VS1053.cpp:150] testComm(): [VS1053] SPI error. SB:804F R1:83FF R2:804F
[ 30457][E][VS1053.cpp:150] testComm(): [VS1053] SPI error. SB:808B R1:8BFF R2:808B
[ 30468][E][VS1053.cpp:150] testComm(): [VS1053] SPI error. SB:809D R1:8AFF R2:809D
[ 30478][E][VS1053.cpp:150] testComm(): [VS1053] SPI error. SB:80D3 R1:89FF R2:80D3
[ 30489][E][VS1053.cpp:150] testComm(): [VS1053] SPI error. SB:80F1 R1:8CFF R2:80F1
[ 30499][E][VS1053.cpp:150] testComm(): [VS1053] SPI error. SB:811B R1:83FF R2:811B
[ 30510][E][VS1053.cpp:150] testComm(): [VS1053] SPI error. SB:8142 R1:8142 R2:8100
[ 30523][E][VS1053.cpp:150] testComm(): [VS1053] SPI error. SB:81C3 R1:89FF R2:81C3
[ 30533][E][VS1053.cpp:150] testComm(): [VS1053] SPI error. SB:81C6 R1:81C6 R2:8900
[ 30544][E][VS1053.cpp:150] testComm(): [VS1053] SPI error. SB:81E1 R1:8CFF R2:81E1
[ 30554][E][VS1053.cpp:150] testComm(): [VS1053] SPI error. SB:81FC R1:81FC R2:8E00
[ 30564][E][VS1053.cpp:150] testComm(): [VS1053] SPI error. SB:822C R1:822C R2:9600
[ 30575][E][VS1053.cpp:150] testComm(): [VS1053] SPI error. SB:8241 R1:90FF R2:8241
[ 30586][E][VS1053.cpp:150] testComm(): [VS1053] SPI error. SB:8283 R1:99FF R2:8283
[ 30595][E][VS1053.cpp:150] testComm(): [VS1053] SPI error. SB:8289 R1:9AFF R2:8289
[ 30605][E][VS1053.cpp:150] testComm(): [VS1053] SPI error. SB:8298 R1:8298 R2:9A00
[ 30615][E][VS1053.cpp:150] testComm(): [VS1053] SPI error. SB:82A1 R1:9CFF R2:82A1
[ 30630][E][VS1053.cpp:150] testComm(): [VS1053] SPI error. SB:837C R1:837C R2:9600
[ 30641][E][VS1053.cpp:150] testComm(): [VS1053] SPI error. SB:83A3 R1:9DFF R2:83A3
[ 31282][I][main.cpp:2836] setup(): [main] Playtask started
[ 31499][I][main.cpp:967] nextPreset(): [main] nextPreset is 6

Edzelf, 2024-01-09:
Send the full logging, then I can see what is wrong.  It is certainly an SPI problem (wiring/configuration).

redstoicescu, 2024-01-10:
Starting ESP32-radio running on CPU 1 at 240 MHz.
[  3042][I][main.cpp:2584] setup(): [main] Version Thu, 26 Dec 2023 14:30:00 GMT.  Free memory 110580
[  3044][I][main.cpp:2585] setup(): [main] Display type is BLUETFT
[  3141][I][main.cpp:2595] setup(): [main] SPIFFS is okay, space 1016801, used 33383
[  3145][I][main.cpp:2616] setup(): [main] Found partition 'nvs     ' at offset 0x009000 with size    20480
[  3147][I][main.cpp:2616] setup(): [main] Found partition 'otadata ' at offset 0x00E000 with size     8192
[  3156][I][main.cpp:2616] setup(): [main] Found partition 'spiffs  ' at offset 0x2F0000 with size  1114112
[  3171][I][main.cpp:2488] fillkeylist(): [main] Read 45 keys from NVS
[  3178][I][main.cpp:1911] readIOprefs(): [main] pin_ir set to 35
[  3179][I][main.cpp:1911] readIOprefs(): [main] pin_enc_clk set to 25
[  3185][I][main.cpp:1911] readIOprefs(): [main] pin_enc_dt set to 26
[  3191][I][main.cpp:1911] readIOprefs(): [main] pin_enc_sw set to 27
[  3197][I][main.cpp:1911] readIOprefs(): [main] pin_tft_cs set to 15
[  3204][I][main.cpp:1911] readIOprefs(): [main] pin_tft_dc set to 2
[  3210][I][main.cpp:1911] readIOprefs(): [main] pin_vs_cs set to 5
[  3216][I][main.cpp:1911] readIOprefs(): [main] pin_vs_dcs set to 32
[  3222][I][main.cpp:1911] readIOprefs(): [main] pin_vs_dreq set to 34
[  3227][I][main.cpp:1911] readIOprefs(): [main] pin_spi_sck set to 18
[  3233][I][main.cpp:1911] readIOprefs(): [main] pin_spi_miso set to 19
[  3240][I][main.cpp:1911] readIOprefs(): [main] pin_spi_mosi set to 23
[  3257][I][main.cpp:2657] setup(): [main] GPIO0 is HIGH
[  3268][I][main.cpp:2657] setup(): [main] GPIO2 is LOW, probably no PULL-UP
[  3278][I][main.cpp:2657] setup(): [main] GPIO4 is HIGH
[  3288][I][main.cpp:2657] setup(): [main] GPIO5 is HIGH
[  3298][I][main.cpp:2657] setup(): [main] GPIO12 is HIGH
[  3308][I][main.cpp:2657] setup(): [main] GPIO13 is HIGH
[  3318][I][main.cpp:2657] setup(): [main] GPIO14 is HIGH
[  3328][I][main.cpp:2657] setup(): [main] GPIO15 is HIGH
[  3338][I][main.cpp:2657] setup(): [main] GPIO16 is HIGH
[  3348][I][main.cpp:2657] setup(): [main] GPIO17 is HIGH
[  3358][I][main.cpp:2657] setup(): [main] GPIO18 is HIGH
[  3368][I][main.cpp:2657] setup(): [main] GPIO19 is HIGH
[  3378][I][main.cpp:2657] setup(): [main] GPIO21 is HIGH
[  3388][I][main.cpp:2657] setup(): [main] GPIO22 is HIGH
[  3398][I][main.cpp:2657] setup(): [main] GPIO23 is HIGH
[  3408][I][main.cpp:2657] setup(): [main] GPIO25 is HIGH
[  3418][I][main.cpp:2657] setup(): [main] GPIO26 is HIGH
[  3428][I][main.cpp:2657] setup(): [main] GPIO27 is HIGH
[  3438][I][main.cpp:2657] setup(): [main] GPIO32 is HIGH
[  3448][I][main.cpp:2657] setup(): [main] GPIO33 is HIGH
[  3458][I][main.cpp:2657] setup(): [main] GPIO34 is HIGH
[  3468][I][main.cpp:2657] setup(): [main] GPIO35 is LOW, probably no PULL-UP
[  3478][I][main.cpp:2657] setup(): [main] GPIO39 is LOW, probably no PULL-UP
[  3479][I][main.cpp:1757] readprogbuttons(): [main] gpio_00 will execute uppreset = 1
[  3482][I][main.cpp:1757] readprogbuttons(): [main] gpio_12 will execute upvolume = 2
[  3489][I][main.cpp:1757] readprogbuttons(): [main] gpio_13 will execute downvolume = 2
[  3497][I][main.cpp:1757] readprogbuttons(): [main] gpio_14 will execute stop
[  3504][I][main.cpp:1757] readprogbuttons(): [main] gpio_17 will execute resume
[  3513][I][main.cpp:2669] setup(): [main] Enable pin 35 for IR
[  3517][I][main.cpp:2674] setup(): [main] Start BLUETFT display
[  4306][I][main.cpp:2280] mk_lsan(): [main] Create list with acceptable WiFi networks
[  4307][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: SSID-11
[  4309][I][main.cpp:2310] mk_lsan(): [main] Added SSID-11 to list of networks
[  4317][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: red
[  4323][I][main.cpp:2310] mk_lsan(): [main] Added red to list of networks
[  4330][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: SSID2
[  4336][I][main.cpp:2310] mk_lsan(): [main] Added SSID2 to list of networks
[  4857][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
[  4953][V][WiFiGeneric.cpp:338] _arduino_event_cb(): STA Started
[  4953][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START
[  5453][I][main.cpp:866] listNetworks(): [main] Scan Networks
[  7377][V][WiFiGeneric.cpp:381] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 10
[  7378][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 1 - SCAN_DONE
[  7382][I][main.cpp:868] listNetworks(): [main] Scan completed
[  7388][I][main.cpp:876] listNetworks(): [main] Number of available networks: 10
[  7395][I][main.cpp:894] listNetworks(): [main]  1 - DIGI-7Mk9                 Signal: -49 dBm, 
Encryption WPA_WPA2_PSK, Not accepttable
[  7407][I][main.cpp:894] listNetworks(): [main]  2 - red                       Signal: -50 dBm, 
Encryption WPA_WPA2_PSK, Acceptable
[  7419][I][main.cpp:894] listNetworks(): [main]  3 - red2                      Signal: -51 dBm, 
[... 159 lines cut ...]
[23715977][I][main.cpp:1296] showstreamtitle(): [main] StreamTitle='Simon & Garfunkel - Old friends / Bookends theme';
[23794575][I][main.cpp:1295] showstreamtitle(): [main] Streamtitle found, 49 bytes
[23794575][I][main.cpp:1296] showstreamtitle(): [main] StreamTitle='NPO Radio 1 - De Nacht van EO - EO';
[24815997][I][main.cpp:1295] showstreamtitle(): [main] Streamtitle found, 37 bytes
[... cut ...]
[28820288][I][main.cpp:4373] gettime(): [main] Sync TOD, old value is 05:23:37
[28820288][I][main.cpp:4375] gettime(): [main] Sync TOD
[28820289][I][main.cpp:4388] gettime(): [main] TOD synced
[30015129][I][main.cpp:1295] showstreamtitle(): [main] Streamtitle found, 38 bytes
[30015129][I][main.cpp:1296] showstreamtitle(): [main] StreamTitle='Beatles - Come together';    
[30265242][I][main.cpp:1295] showstreamtitle(): [main] Streamtitle found, 45 bytes
[30265243][I][main.cpp:1296] showstreamtitle(): [main] StreamTitle='NPO Radio 1 - Vroeg! - BNNVARA';

Edzelf, 2024-01-10:
It is an SPI problem.  At [ 19661] you can see that the SPI test succeeded with a low SPI frequency (0.2 MHz) succeeded.  The test with a high SPI frequency (5 MHz) fails.  So first thing to do is to check the SPI bus wiring. 
At [ 20376] you can see that the chip in your soundmodule is a VS1001 instead of a VS1053.  It might also be a fake VS1053.  A real VS1001 will work for MP3 data only.
You may try to remove line 205 in VS1053.cpp ( `S1053_SPI._clock = 5000000 ;     // SPI Clock to 5 MHz.` ), but it would be better to replace your VS1053 module with a better one.

redstoicescu, 2024-01-10:
changed in config.h  dec1003 to 1053 :
[ 84816][W][VS1053.cpp:162] testComm(): [VS1053] This is not a VS1053, but a VS1001 instead!

redstoicescu, 2024-01-10:
modified vs1053
rebooting continuous

redstoicescu, 2024-01-11:
Solved , wiring problem 
Thank you for your patience