Edzelf/ESP32-Radio #531: Arduino IDE compiling error in aac_decoder.cpp library

amishrabbit, 2024-07-08:
Hello there. While attempting to build/upload to a "Hackerbox #70" hardware build (https://www.instructables.com/HackerBox-0070-Radio-Over-Internet/) of this project, the Arduino IDE 1.8.19 throws an error with this detail:

ESP32Radio\aac_decoder.cpp: In function 'void GenerateNoiseVector(int*, int*, int)':

aac_decoder.cpp:5297:41: error: invalid conversion from 'uint32_t*' {aka 'long unsigned int*'} to 'unsigned int*' [-fpermissive]

 5297 |         coef[i] = ((int32_t)Get32BitVal((uint32_t *)last)) >> 16;

      |                                         ^~~~~~~~~~~~~~~~

      |                                         |

      |                                         uint32_t* {aka long unsigned int*}

As an inexperienced developer who is just trying to follow directions, this has been a challenging project to complete. I'd welcome any guidance or assistance you can provide to work around this problem.

Thanks!

Edzelf, 2024-07-09:
Try this:
coef[i] = ((int32_t)Get32BitVal((unsigned int*)last)) >> 16;

Better use the new version of the ESP32 radio, see [here](https://github.com/Edzelf/ESP32Radio-V2).

amishrabbit, 2024-07-09:
Thanks for the update, Ed. I really appreciate it. 

I was having difficulty with reflashing this particular kit, which is a few years old now. I wanted to reproduce the steps I went through originally, but so many of the libraries have updated beyond supporting what they did a few years ago, it was very, very difficult. 

I was also trying to avoid installing an entirely different IDE other than Arduino, which I am comfortable with. 

In the end, the trick that seemed to work for the Hackerboxes kit was to flash the settings into the device using the V1 [Esp32_radio_init.ino](https://github.com/Edzelf/ESP32-Radio/blob/master/tools/Esp32_radio_init.ino) in Arduino IDE with the "Erase entire flash memory" enabled, then switching to the Visual Studio Code/PlatformIO to flash over the (V2) firmware. That managed to get the user interface working and the device bootable. 

Without flashing the settings first in Arduino, this particular device build (ESP-WROOM32 "DOIT ESP32 DEVKIT V1" + VS1053 + ST7735R 'GreenTab') does not seem to "take" the defaultsettings.txt content that was being flashed from the V2 software over PlatformIO, and fails to boot at all. Doing it this way, the device fires right up.

So while I'm grateful for the line of updated code, it was not necessary.

However, I'm now experiencing a different problem with this hardware so I will close this ticket and open a new one for that issue.

Thanks again!