Edzelf/ESP32-Radio #533: cannot compile -  fatal error: utils.h:

mickeypop, 2024-10-20:
REF  Arduino IDE

**ERROR  ...\ESP32-Radio.ino:144:82: fatal error: utils.h: No such file or directory**

even with debugging set to VERBOSE, all I get is utils.h is missing

what lib uses utils.h so I can add it and fix this

Edzelf, 2024-10-21:
I probably uploaded the V2 version to this repositry.   I have corrected this.  But please use the V2 version.  It it much better.

mickeypop, 2024-12-17:
the  error 
C:\Users\micke\AppData\Local\arduino\sketches\FA72BF97596CD6E716FD829CDF28CBCE\sketch\ESP32-Radio.ino.cpp -o nul
Alternatives for utils.h: []
ResolveLibrary(utils.h)
  -> candidates: []
C:\Users\micke\Documents\Arduino\CUSTOMERS\PersonalProjects\ESP32-Radio\ESP32-Radio.ino:144:10: fatal error: utils.h: No such file or directory
  144 | #include "utils.h"                                        // Some handy utilities
      |          ^~~~~~~~~
compilation terminated.
exit status 1

Compilation error: utils.h: No such file or directory

What library is utils.h from?

Edzelf, 2024-12-18:
If you are stil using the V1 version: there is no include for "utils.h".  Your main program is the wrong version.
For version 2, the utils.h is in the include directory.